AstarVienna/irdb #10: Broken TER_armazones_default_NIR_IMG.dat

hugobuddel, 2021-03-20:
TER_armazones_default_NIR_IMG.dat is broken:

https://github.com/astronomyk/irdb/blob/924d212d30879eeaed14afdcd1d6d2a278d9a4d8/Armazones/TER_armazones_default_NIR_IMG.dat#L154

See
https://gitlab.astro-wise.org/micado/micadowise/-/issues/138
https://gitlab.astro-wise.org/micado/micadowise/-/merge_requests/189
https://gitlab.astro-wise.org/micado/micadowise/-/jobs/31298

hugobuddel, 2021-03-20:
Seems like the first line of that table was appended to this line instead of removed: https://github.com/astronomyk/irdb/commit/51f8aa610bf156704f11f08c70e5d31aedc6e14b#diff-f585158fdbaf36e1d8a93d0635e16ebc099e862704716098510c94f19cef49ba

miguelverdugo, 2021-03-22:
offended line deleted

hugobuddel, 2021-03-22:
> offended line deleted

The line seems to still be there. I've already submitted a pull request (#11) to fix it, all you have to do is press the "Merge pull request" button.

closed 2021-03-22 by miguelverdugo

hugobuddel, 2021-03-22:
Thanks @miguelverdugo ! @astronomyk can you update the Armazones package on the website? (In case that is not yet automatic.)

PR #11 (2021-03-22): Fix broken line
Remove the accidentally copied line.
Closes #10
commit 1733c558 (2021-03-22): Fix broken line

Remove the accidentally copied line.
Closes #10
PR #15 (2021-03-30): Fix broken line
Remove the accidentally copied line.
Closes #10