AstarVienna/irdb #12: Add CI to at least have astropy read in all tables?

hugobuddel, 2021-03-20:
Perhaps we can add some continuous integration tests that read in all the tables?

That would have caught #10 right of the bat!

oczoske, 2021-03-22:
I have added such tests in [test_metis.py](https://github.com/astronomyk/irdb/blob/master/METIS/tests/test_metis.py) (yamls, filters, ter curves, fpa layout, linearity). There's quite a bit of redundancy between the file types but I wanted to keep them separate. Maybe you want to adapt that to MICADO?  Suggestions for improvements are welcome.

hugobuddel, 2021-03-23:
Thanks I'll look into that.

It seems that it are the MICADO tests that prevent the CI from succeeding, so fixing those would probably already be useful.

I might be able to look into that next month; I didn't really have these tests on my radar and now it bites me.