AstarVienna/irdb #14: User manual example D

oczoske, 2021-03-30:
https://github.com/astronomyk/ScopeSim_FDR_Docs/blob/master/MICADO_ScopeSim_user_manual/tex/D_basioc_spectroscopy_with_micado_sci_spec.pdf

Package `MICADO_Sci` does not include any filters curves. Hence, `instruments/MICADO.zip` needs to be downloaded as well.

astronomyk, 2021-03-30:
> :
> In the final plot (In [36]), mismatch between waves and flux. There are
> 999 fields of view. The image has size 1000 in the spatial direction, and
> 800 in the wavelength direction. Also, my image looks different:
> [image: Figure_3]
> <https://user-images.githubusercontent.com/16150218/113002049-b6d9ef80-9171-11eb-965a-43612da3e8de.png>

Interesting .....
(not necessarily the good type of interesting :(
-------------------------------------
Kieran Leschinski
Department of Astrophysics
University of Vienna

teutoburg, 2025-09-10:
@oczoske is this still relevant? Close?