AstarVienna/irdb #28: Ensure Focal Plane Array (FPA) parameters are up to date

hugobuddel, 2022-01-28:
Ric created a [technical note about some the Focal Plane Orientation](https://bscw.mpe.mpg.de/bscw/bscw.cgi/5161261?op=preview) (ELT-TRE-MCD-56300-0132, v1.3, dated May 6th 202), that seems to not be incorporated in [MICADO/FPA_array_layout.dat](MICADO/FPA_array_layout.dat), which is from October 2020.

It is not entirely clear what needs to change (if anything indeed needs to change).

@astronomyk can you verify that the MICADO package is up to date with the above note?

astronomyk, 2022-01-28:
> incorporated in MICADO/FPA_array_layout.dat, which is from October 2020.

Thanks for reminding me of the document. I haven't read it, so I can not
confirm if the ScopeSim Micado package is up to date. I'll add this to my
to-do list for when I'm back at work.

hugobuddel, 2022-02-10:
Closed by #32

closed 2022-02-10 by hugobuddel