AstarVienna/irdb #56: Reformat and add new MAORY PSFs to IRDB

astronomyk, 2022-04-08:
All the new PSFs (April 2022) are available on the univie server: https://www.univie.ac.at/simcado/InstPkgSvr/psfs/new_maory_psfs/

**Note: DO NOT add these fits files to the IRDB. THEY WILL BE TOO LARGE FOR GIT**

Tasks:

- [ ] Make a FITS data file for a FieldConstantPSF object:
  - File format documentation is here: https://github.com/AstarVienna/ScopeSim/blob/master/docs/source/interfaces/file_formats/psfs.rst#spatially-constant-psfs
  - [ ] Shrink the PSFs to 2x2 arcsec
  - [ ] Combine the PSF images into a HDUList with a new ImageHDU extension for each wavelength
  - [ ] Add WAVE0 and WAVELENG to the header of each ImageHDU with the wavelength in um
  - [ ] Make versions of this object with:
    - [ ]  the original 0.75mas pixel scale, 
    - [ ] a down-sampled 4mas pixel scale (use scipy.zoom() for this)

- [ ] Make a FieldVaryingPSF object
  - File format description: https://github.com/AstarVienna/ScopeSim/blob/master/docs/source/interfaces/file_formats/psfs.rst#field-varying-psfs 
  - The same as for the FieldConstantPSF object above, but
  - [ ] combine the 9 different positions for each wavelength into a FITS cube and add these cubes as ImageHDUs to a HDUList in exts 2-10
  - [ ] add a TableHDU to ext 1 with the x,y positions for each cube layer - NOTE the coordinates of each layer should be the same for each cube 

- [ ] Find a place to keep these large files, as they **SHOULD NOT BE ADDED TO THE IRDB**
- [ ] Add the relevant effects to the MAROY/MAORY.yaml file in the irdb.

carmeloarci, 2023-12-14:
Waiting for the TIP TOP semi-automatic PSF generation for MORFEO-MICADO,  maybe it is possible to make available the PSF files from somewhere else.

hugobuddel, 2023-12-18:
In #58 @carmeloarci says

> I already passed the information, however, the latest release of the field constant psf is available at the link: [MORFEO "PSF 2023"](https://drive.google.com/drive/folders/1-WxTTyrQ5oZVxO420UuNkYRsjeuQe5D0?usp=sharing)

Those files are from July 2023, so maybe we should update the files in https://scopesim.univie.ac.at/InstPkgSvr/psfs/new_morfeo_psfs/ConstPSFs/ , and/or provide an easy way for users to specify their own set of files.

(And perhaps we should not use `new` in the names of directories.)