AstarVienna/irdb #58: MICADO Check that PSFs are scaling with wavelength

AstarVienna, 2022-04-11:
From JvdB

"It is not obvious to me that (or how?) the PSF changes when changing filters. I would expect that the PSF has a different size in K-band compared to J-band. I know anisocado can do this properly, but I guess by default a fixed PSF is used."

While we're at it, we could add a page to the docs showing what's going on here

yosata, 2023-12-13:
Being reminded of this issue after looking at the development dashboard today, I remembered that I got this question answered myself when designing the cover for my thesis.

Today I quickly checked if everything still works as expected in the latest version of ScopeSIM.

In summary; yes the PSFs are scaling as expected with wavelength. See the attached Jupyter Notebook. Maybe someone else can copy/paste or modify this for the documentation.

[ScopeSIM PSF scaling verification.pdf](https://github.com/AstarVienna/irdb/files/13662692/ScopeSIM.PSF.scaling.verification.pdf)

Uhm... apparently I don't know how to add an .ipynb file to this comment. Send me an email or message on slack and I can send it directly.

carmeloarci, 2023-12-14:
I was inspired by Joost' notebook and I ran a similar one using MORFEO MCAO mode instead of SCAO, finding also a way to load the latest MORFEO PSFs. 
[MORFEO sims.pdf](https://github.com/AstarVienna/irdb/files/13676799/MORFEO.sims.pdf)
Actually, in the MORFEO setup currently available for MORFEO-MICADO the I band PSF is not available, and PSF does not scale with wavelength.

carmeloarci, 2023-12-14:
I already passed the information, however, the latest release of the field constant psf is available at the link: [MORFEO "PSF 2023"](https://drive.google.com/drive/folders/1-WxTTyrQ5oZVxO420UuNkYRsjeuQe5D0?usp=sharing)

hugobuddel, 2023-12-18:
@yosata thanks for the update! Perhaps good to keep this issue open to put the information in the documentation.

@carmeloarci I missed your earlier update about the PSFs. Should we make an issue for updating the MORFEO PSFs?

hugobuddel, 2023-12-18:
> @carmeloarci I missed your earlier update about the PSFs. Should we make an issue for updating the MORFEO PSFs?

I suppose that would be #56