AstarVienna/irdb #77: [MICADO] Weird bend in HK Spec traces

astronomyk, 2022-04-27:
Possibly due to the 4th order polynomial in `effects.spectral_trace_utils.Transform2D`
Look into removing this

oczoske, 2023-07-24:
[image]

The figure shows a simulation of a MICADO long-slit observation (15000x20) in HK with a regular line list (done by Joshua Jost), with the control points defining Order 3_1 overlaid. The simulation clearly displays the curvature in the lines, which looks consistent with the control points. The curvature appears strongest to the left of the left control point, where the polynomial extrapolates. The control points lie at xi = (-5, -1.5, 0, 1.5, 5, 10) arcsec from right to left. The slit if defined to lie between -1.5 and 13.5 arcsec, so there is a mismatch between the slit position and the control points.
To do: Check that this slit position is in fact correct. Moving the slit to smaller xi would centre the trace better on the detector (as for Order 4_1, visible in the overview window top right).

Lowering the order of the fit in the xi direction would reduce the curvature but requires a bit of work (fourth order is still required in the wavelength direction). A wish for more control points is unlikely to be fulfilled quickly...

oczoske, 2023-07-24:
Ric Davies has confirmed that the slit position is in fact from -5 to +10 arcsec.

PR #123 (2023-07-24): x position of long slit corrected
As confirmed by Ric Davies, the long slit should be positioned between -5 and +10 arcsec. This closes #77 .
closed 2023-07-24 by oczoske