AstarVienna/irdb #105: Can't download Paranal package

Isa29bella, 2023-06-13:
Hi,
I tried to download the Paranal location package to run a simulation with HAWKI. The VLT and HAWKI packages downloaded fine, as did the Armazones, ELT and MICADO packages I used previously. However, for the Paranal package, I get the following error: `HTTP Error 404: Not Found`

Here is the most basic version of what I tried to do:
```
import scopesim as sim
sim.download_packages(["Paranal"])
``` 

Thanks :)

hugobuddel, 2023-06-14:
Thanks, something seems to have gone wrong with uploading the package. I'll upload the package later today.

In the meantime, you could perhaps use an older version of the package with:
```python
import scopesim as sim
sim.download_packages(["Paranal"], release="2021-12-01")
```

hugobuddel, 2023-06-14:
The latest Paranal package is uploaded, so you should be able to download the file as normal. Could you please close the issue if everything indeed works as expected?

I do think I've identified the main cause of the problem (see later issue), so hopefully we can prevent this from happening again.

Isa29bella, 2023-06-14:
Indeed the download works now, thanks a lot!
However, when I try to use HAWKI (which was my original goal), I now get a new error:
Running:
```
sim.UserCommands(use_instrument="HAWKI")
```
Produces:
```
ParserError: while parsing a block mapping
  in "C:\Uni\Astro\10Semester\ELT\inst_pkgs\HAWKI\HAWKI.yaml", line 1, column 1
expected <block end>, but found '<block sequence start>'
  in "C:\Uni\Astro\10Semester\ELT\inst_pkgs\HAWKI\HAWKI.yaml", line 31, column 2
```
Should I open a separate issue for that and close this one? As far as I can tell this looks like an unrelated problem...
Sorry to bother you with all of this.

hugobuddel, 2023-06-14:
No problem, these things should just work, so thanks for reporting them!

Apparently there was a similar issue with the HAWKI package (bit different), so I've released a new HAWKI package. The VLT package was all right.

Note that what you can also do is clone this repository, and then let ScopeSim search that path. E.g.:
```
sim.rc.__config__["!SIM.file.local_packages_path"] = "/path/to/irdb/clone"
```

Nevertheless, we should have done a better job.

closed 2023-06-14 by Isa29bella