AstarVienna/irdb #107: Publishing packages.yaml together with packages does not work when using multiple branches.

hugobuddel, 2023-06-14:
Today there was an issue (#105) that it was not possible to download the Paranal package.

That was because https://scopesim.univie.ac.at/InstPkgSvr/packages.yaml listed `Paranal.2022-04-09` as latest (and stable) version, but that package was not available in https://scopesim.univie.ac.at/InstPkgSvr/locations/ .

`Paranal.2022-04-09.zip` was probably never uploaded. This might not have been problematic, if indeed `packages.yaml` would have pointed to the correct file, which might have been the case for a while. However, I 'recently' (March) uploaded a new MICADO package. That publishing would also have uploaded `packages.yaml`, thereby breaking the Paranal download (for 3 months without anyone telling us). Not sure this was indeed the case though, as older version of `packages.yaml` are not kept on the server.

However, a similar problem is occurring with MAAT/OSIRIS. [`packages.yaml` in the `dev_maat` branch](https://github.com/AstarVienna/irdb/blob/dev_maat/irdb/packages.yaml) contains
https://github.com/AstarVienna/irdb/blob/23f7fba051b5bf8259d0417f3318a69861a7576b/irdb/packages.yaml#L45-L49

while [`packages.yaml` in the `master` branch](https://github.com/AstarVienna/irdb/blob/dev_maat/irdb/packages.yaml) (that I uploaded when uploading MICADO) contains
https://github.com/AstarVienna/irdb/blob/143264e8c1077265049a59f3f691a7719031e7ce/irdb/packages.yaml#L45-L49

Only `OSIRIS.2023-01-13.zip` exists on https://scopesim.univie.ac.at/InstPkgSvr/instruments/ , but I overwrote the `packages.yaml` from the `dev_maat` branch, so now it is not possible anymore to download OSIRIS.

GTC and LaPalma are also broken, see https://github.com/AstarVienna/ScopeSim/pull/225 .

I'm too wary right now to think about the proper solution to this problem. Some ideas, in order of increasing usefulness and difficulty:

* Easy: Only publish packages from the `master` branch (or perhaps the `dev_master` branch for development packages). This would only require changes in our workflow, not in the code.
* Medium: Get rid of `packages.yaml`. The filenames (e.g. `test_package.2022-04-27.dev.zip` contain all the relevant information already, so no need to consult `packages.yaml` at all. This requires code change though.
* Hard: Create PyPI packages of all the packages in the IRDB. Then we get this versioning stuff for free. The packages are all rather small in size anyway. I like this the most, but it is the most work.

@teutoburg what do you think?

hugobuddel, 2023-06-14:
To clarify, it was trivial to resolve the Paranal problem (#105) simply by uploading it again from the master branch, because the Paranal package is the same in all branches. However, solving OSIRIS is harder. The four options are:

1. Upload OSIRIS from the master branch. This would lead to an outdated version on the server.
2. Upload OSIRIS from the dev_maat branch. This would break all other packages like MICADO, because the `packages.yaml` from the dev_maat branch would be used.
3. First merge dev_maat into (dev_master into) master, before doing 1 above. This would probably cause problems too, because if this merging would have been possible, it would probably have been done when OSIRIS.2023-01-13  was released.
4. Merging master into dev_maat, before doing 2 above. This would then have to be done every time any package is released, and cannot be done if there are more such dev_instrument branches.

teutoburg, 2023-06-14:
Regarding the OSIRIS and dev_maat issue, it seems if we go for the *medium* option, then the OSIRIS option no. 2. would work? Or indeed, if `packages.yaml` would not update **every** package version number.
If we do upload OSIRIS from dev_maat, it should probably be considered a `latest` version ending on `.dev`, while the `stable` version is the (outdated) one from the master branch.

teutoburg, 2023-06-14:
While we talk about all of these version issues, what do you think about using *sematic versioning* for IRDB instead of the date?

teutoburg, 2023-06-14:
Considering these things, I do agree about the versioning scheme. But it's good to remind ourselves why we do things the way we do them :)

If I would have created the code to download these things from the server, I probably would have either gone with hardcoded paths as well, or with (the url version of) glob. But the distinction in `locations`, `telescopes`, `instruments` makes sense IMO, adding some meaningful structure. If we want, we might include a way to "list locations", "list telescopes" or "list instruments" in the future, just as conveniences for the enduser. Or e.g. "list all available instruments for this telescope" or "list all available telescopes for this location", or a tree-like combination. Just a few ideas that popped into my head :)

closed 2023-06-22 by teutoburg