AstarVienna/irdb #141: MICADO limited magnitude comparison with Davies+2010

hugobuddel, 2023-11-06:
Based on slack discussion with Francesco Valentino

> I’ve started looking into how PSFs are currently implemented in MICADO simulations in ScopeSim.
> The conversation stemmed from my attempt to reproduce something similar to _get_limiting_mags in SIMCADO and compute limiting magnitude for SCAO, MCAO simulations in ScopeSim. Since I get some different results from Davies+2010, we were wondering if this is due to the background, PSF, or other factors.
> 
> When I implement some simple Micado simulations with micado = sim.OpticalTrain(micado_cmds), I can see that the code calls for MORFEO.yaml (MCAO) and MICADO_Standalone_RO.yaml (SCAO). There I can see two non-varying PSFs that are used:
>
> ```yaml
> -  name : relay_psf
>   description : SCAO PSF
>   class : FieldConstantPSF
>   kwargs:
>     filename : PSF_SCAO_ConstPSF_0_5off.fits
>     warning : “Default PSF is NOT field varying. See documentation.”
> ```
> and
> ```yaml
> -  name : morfeo_generic_psf
>   description : MORFEO field varying MCAO PSF
>   class : FieldConstantPSF
>   kwargs:
>     filename : PSF_MCAO_ConstPSF_40_18_6.fits
>     warning : “Default PSF is not Field Varying. See Documentation”So here’s some questions:
> ```
> -    How were these PSFs generated? With Anisocado? In the header of the psf fits files, I can read some of the parameters (e.g., for PSF_MCAO_ConstPSF_40_18_6.fits: wavelength=1.2 (um), EsoMedian (atmospheric conditions from Anisocado?), Seeing=0.67, Strehl = 0.047, etc.)
> -    How’s the wavelength dependence accounted for? It’s an analytical rescaling?
> -    Looking at Anisocado, one can create a different PSF and pass it on in fits format, I think. Does one have to add the information to a yaml file or can the psf be passed on directly within python, after creating the PSF?
> -    Is .fits format from Anisocado the standard that will be kept? I ask because other groups are working on realistic PSFs that depend on the availability of nearby stars, distance from the target, etc. We were wondering if / how these PSFs can be implemented in future simulations.
> 
> 
>
> For reference, here’s the plot I was trying to recreate from Davies+10
> [image]
>
> And here’s my version
> [image]
>
> Davies+10 numbers (AB):
>[image]

hugobuddel, 2023-11-06:
I don't actually know the answers to these questions directly, @teutoburg or @astronomyk can you chime in?

Nevertheless, my attempt:
-    I do believe the SCAO PSF is generated with AnisoCADO, but not sure. The MCAO PSF is not I think.
-    As for the wavelength dependence, there are multiple FITS extensions for different wavelenghts, but not sure how it is interpolated. Could be just nearest neighbour
-    I think it should be possible to just use the AnisoCADO PSFs directly, as in, the filename of the FieldConstantPSF class can be overwritten
-    We don't have any plans for AnisoCADO or the file format at the moment. You are suggesting to stabilize it so other tools can use it too?

As for the limiting magnitude, perhaps we should first try to verify the results that Francesco got.

teutoburg, 2023-11-06:
Answers from @astronomyk :

> * I do believe the SCAO PSF is generated with AnisoCADO, but not sure. The MCAO PSF is not I think.

By default, the SCAO PSF is from a static file, which was generated using AnisoCADO. The MCAO PSF is also generated fom AnisoCADO using static strehl numbers (K,H,J=40,15,7%). By default, both SCAO & MCAO PSFs are read from static FITS files.

> * As for the wavelength dependence, there are multiple FITS extensions for different wavelenghts, but not sure how it is interpolated. Could be just nearest neighbour

"Could be just nearest neighbour" - correct

The PSF FITS extensions are chosen as the midpoint for each broadband filter.

> * I think it should be possible to just use the AnisoCADO PSFs directly, as in, the filename of the FieldConstantPSF class can be overwritten

In principle, this is possible and in theory straightforward (quote by @astronomyk ). See [the docs](https://scopesim.readthedocs.io/en/latest/reference/scopesim.effects.psfs.html#scopesim.effects.psfs.AnisocadoConstPSF).

> * We don't have any plans for AnisoCADO or the file format at the moment. You are suggesting to stabilize it so other tools can use it too?

For the moment we're using FITS files, but we're open to creating an effect which uses numpy arrays as an input, for example...

carmeloarci, 2023-11-06:
Hi, for the "limiting magnitude comparison" it is essential to cross-check that Davies+2010 and ScopeSim use the same input, the essential ingredient being:
- Sky Background
- Transmissivity/quantum efficiency of the system  telescope-optical relay (MICADO stand-alone or Morfeo)-MICADO-Detector
- Detector Dark/RON
- PSF

As reminder. The latest MCAO PSF are here: [https://drive.google.com/drive/folders/1-WxTTyrQ5oZVxO420UuNkYRsjeuQe5D0?usp=drive_link](https://drive.google.com/drive/folders/1-WxTTyrQ5oZVxO420UuNkYRsjeuQe5D0?usp=drive_link)

We need to define/check the Signal to Noise ratio definition
- area of the core of the PSF used for the computation of the signal
- the way the noise is estimated

For SIMCADO the function _get_limiting_mag is described here:
[https://simcado.readthedocs.io/en/latest/reference/simcado.simulation.html](https://simcado.readthedocs.io/en/latest/reference/simcado.simulation.html)
and this computes the signal from the measurements and the noise using basic principles:
[image]
The signal is computed on the simulated image of a grid of equal magnitude stars, by computing the total of the counts found within an aperture (i.e. 4x4 or 2x2 pixel centered on the star peak).

From an observational point of view, it is more practical to compute both signal and noise for the detection of a point source from the image (background subtracted image). In the case (as in the grid of stars) you have multiple identical sources, you may fix an aperture size and on this aperture compute the average of the total of the counts (this is the signal), while the noise is the sdev of the total of the counts you measure on an area of detector free of light sources.

hugobuddel, 2023-11-27:
Attached a notebook by Francesco to produce the figures. Based on https://nbviewer.org/github/astronomyk/SimCADO/blob/master/docs/source/_static/python_notebooks/Limiting_Magnitudes.ipynb

[test_performance.zip](https://github.com/AstarVienna/irdb/files/13477219/test_performance.zip)

Produced final figure:
[image]

In code:
```python
# # Test on MICADO performances
# (F. Valentino, ESO)
#
# See the example in SimCado: https://nbviewer.org/github/astronomyk/SimCADO/blob/master/docs/source/_static/python_notebooks/Limiting_Magnitudes.ipynb

# +
import numpy as np
import matplotlib.pyplot as plt
from matplotlib.colors import LogNorm
import matplotlib.ticker as mtick

# Update plot parameters
plt.rcParams.update({'font.size': 14, 'axes.labelsize': 14, 'axes.linewidth': 2, 
                     'xtick.minor.width': 2, 'xtick.major.width': 2, 'ytick.major.width': 2, 
                     'ytick.minor.width': 2})

from astropy import units as u
from sklearn.linear_model import LinearRegression
from photutils.aperture import CircularAperture, CircularAnnulus, aperture_photometry
from scipy import interpolate
from scipy.optimize import curve_fit

# ScopeSim
import scopesim as sim
import scopesim_templates as sim_tp
import os

from copy import deepcopy

# This is needed to download the ELT packages - only once in the selected folder
# (this line can be commented out after the first usage):
if os.path.isdir("inst_pkgs") == False:
    sim.download_package(["instruments/MICADO", "instruments/MORFEO", "telescopes/ELT", "locations/Armazones"])
# -

import numpy as np
def get_snr_aperture_photometry(image, x_pos, y_pos, circle_radius, annulus_inner_radius, annulus_outer_radius):
    
    """
    Perform aperture photometry to calculate signal, noise, background, and SNR values for given positions.
    
    Parameters:
        image (numpy.ndarray): The input image.
        x_pos (array-like): X-coordinates of aperture positions.
        y_pos (array-like): Y-coordinates of aperture positions.
        circle_radius (array-like): Radii of circular apertures.
        annulus_inner_radius (array-like): Inner radii of annuli for background estimation.
        annulus_outer_radius (array-like): Outer radii of annuli for background estimation.
        
    Returns:
        tuple: A tuple containing lists of calculated values (flux, err, std, snr_values, sky).
    """
    
    # Initialize lists to store calculated values
    flux = []
    err = []
    std = []
    sky = []
    snr_values = []
    
    i = 0  # Initialize a counter for iterating through positions
[... 223 lines cut ...]
ax.set_title(mode_name[0]+' '+mode_name[1]+' - '+refmag)

# Manage ticks:
ax.yaxis.MinorTick='on'
ax.xaxis.MinorTick='on'
ax.yaxis.set_minor_locator(mtick.AutoMinorLocator(10))
ax.xaxis.set_minor_locator(mtick.AutoMinorLocator(10))
ax.tick_params(which='minor', length=3, direction='in')
ax.tick_params(which='major', length=6, direction='in')

# Save the figure:
fig.tight_layout();
fig.savefig('plots/performance_micado_'+mode_name[0]+'_'+mode_name[1]+'_'+refmag+'.png', format='png')
# -

# This is the plot in Davies+2010:

# ![Screenshot 2023-08-23 at 16.46.34.png](attachment:08459127-1c22-49f3-a020-8f6e95b65180.png)![Screenshot 2023-08-23 at 16.46.41.png](attachment:94933064-259a-4671-80a0-e0bcb5561ee7.png)

# +
#for i in micado.effects: print(i)
# -

micado.cmds
```