AstarVienna/irdb #151: MICADO MCAO Configuration should not include the Dichroic

carmeloarci, 2024-01-29:
I see that  the file:
LIST_MICADO_mirrors_static.dat
includes the row
I09_SCAO_PO 0.5     0.0     45      -190            transmission   TER_SCAO_dichroic.dat

This element should appear only in the SCAO configuration and not in the MCAO.
I suggest to remove it from the file above 
And possibly to include it somewhere specific to the SCAO as the file LIST_RO_SCAO_mirrors.dat 

This element blocks the I-band light.

astronomyk, 2024-02-13:
Hi @carmeloarci 

Thanks for the heads up. I agree with the change you're proposing. 
Just for the sake of including a reference for the change in the file meta data - do you have a document number to refer to which details the current optical designs of both MICADO and MORFEO, or just the SCAO system really?

Thanks!

carmeloarci, 2024-02-13:
Ciao @astronomyk 
for the optical design (in particular of the Optical Relay), you may refer to FDR documentation(s):
************************************************
ELT - MICADO Phase C
MICADO Relay Optics Design and Analysis Report
Doc. No.: ELT-TRE-MCD-56301-0028
************************************************
OFDR MORFEO Analysis Report 
Document Number: E-MAO-SF0-INA-ANR-001
Document Version: 01
Released On: 2023-06-21
************************************************
ciao

carmeloarci, 2024-02-23:
Did you receive my pull request (as of today) on this topic?

hugobuddel, 2024-02-23:
Hi @carmeloarci , thanks for moving the dichroic! There doesn't seem to be a pull request yet though.

It seems that this is the commit you want to merge: https://github.com/AstarVienna/irdb/compare/dev_master...carmeloarci:irdb:dev_master . There is a "Create pull request" button (for me at least); shall I create a PR, or do you want to do that?

By the way, you committed to your own `dev_master` branch, which should work fine for this particular PR, but could become annoying later if the 'upstream' (AstarVienna) `dev_master` gets the commits in a different order, because then you'd have to juggle the branches being inconsistent. It is usually easier to make pull requests from a 'feature branch'.

closed 2024-02-28 by hugobuddel