AstarVienna/irdb #162: Improve documentation on RTD, especially for METIS

teutoburg, 2024-03-11:
This should be ready in time for the METIS science team meeting.

- [ ] Improve the look and feel of the IRDB RTD site.
- [ ] Make sure all notebooks are included (e.g. IFU).
- [ ] Make sure all notebooks are run, i.e. there is actually output shown on RTD.
- [ ] Make sure all information is correct / up to date.
- [ ] Check #131 
- [ ] Generally reconsider RTD structure, it's currently limited to METIS and MICADO.

hugobuddel, 2024-03-12:
The notebooks probably cannot be ran on RTD itself, as they probably require too much RAM. But for METIS we can try to do everything on RTD; MICADO is more problematic IIRC. In the long run it might be better to build the documentation on zeus and then upload it to RTD, assuming that is possible.

teutoburg, 2024-03-12:
> In the long run it might be better to build the documentation on zeus and then upload it to RTD, assuming that is possible.

Yeah, that was my thought as well. I'd like to explore that option at some point in the not-too-distant future...