AstarVienna/irdb #163: Update to METIS LSS trace descriptions

astronomyk, 2024-03-20:
Here is the latest (final) version of the LSS trace descriptions from MPIA. 

@wkausch  - once these are in the METIS package, you should be able to restart your LSF analyses.

[2023-06-20-spectra_Grism-L_band.txt](https://github.com/AstarVienna/irdb/files/14665143/2023-06-20-spectra_Grism-L_band.txt)
[2023-06-20-spectra_Grism-M_band.txt](https://github.com/AstarVienna/irdb/files/14665145/2023-06-20-spectra_Grism-M_band.txt)
[2023-06-20-spectra_Grism-N_band.txt](https://github.com/AstarVienna/irdb/files/14665146/2023-06-20-spectra_Grism-N_band.txt)

Except from MPIA email with extra info:
```
...

Some notes (that I have also included in the txt files):

*Notice that this model includes the manufacturing tolerances of the grisms, then the spectra length is a little bit smaller that the ideal length.
That are:
L-band, 2.9-4.2 microns: ideal 1920 px, designed 1760 px.
M-band, 4.5-5.2 microns: ideal 1920 px, designed 1680 px.
N-band, 7.6-13.4 microns: ideal 1984 px, designed 1862 px.

* the x,y [mm] coordinates on the detector have now the origin of coordinates at the bottom-left corner of the detector sensitive surface.
In this way I think there will not be ambiguity in the future (I made this because zemax made some inversions, so it is clear if I put the coordinates wrt to a physical reference on the detector).
For the three grisms, at the detector, we will find the short wavelengths (2.9, 4.5 or 7.6 microns) at the top of the detector, and the long wavelengths (4.2, 5.2 or 13.4) at the bottom.
This is the final configuration decided.
BTW, now the coordinates have 6 decimals! Tell me if this is ok.

* the slit at CFO FP2 is now on-axis.
 (this was decided some time ago, the slit is NOT anymore decentered!!)
However, remember that the detector is shifted (in LM detector in Y direction by 32 px negative, and in N detector in Y direction  by 52.5 px negative). Then, the grisms are optimized to obtain that decenter Y of the spectra on the detector.

...
```

wkausch, 2024-03-20:
@astronomyk Many thanks for this info, sounds very well. Is it possible to implement this by 5th April 2024? I'd like to come to Vienna from 8th Apr onwards for the scheduled Hackathon to create the simulations required for the answering of the FDR AIs (the week after is a conference in Vienna where I'm attending, so combining it with the hackathon is easily possible).

teutoburg, 2024-04-11:
Closed by #168

closed 2024-04-11 by teutoburg